clearpathrobotics/clearpath_simulator #11: Robot meshes and structure are not correctly shown

fmrico, 2023-08-09:
- OS: Ubuntu 22.04)
 - ROS 2 Distro: Humble
 - Built from source
 - Package version: 
   - clearpath_common: humble
   - clearpath_config: main
   - clearpath_robot: main
   - clearpath_simulator: main
   - Ignition Gazebo, version 6.14.0

 **Expected behaviour**
I am following the documentation at: https://docs.clearpathrobotics.com/docs/ros/config/yaml . My robot.yaml is:

```
serial_number: a200-0000
version: 0
system:
  hosts:
    self: argo
    platform:
      argo: 127.0.0.1
    onboard: {}
    remote: {}
  ros2:
    username: fmrico
    namespace: /
    domain_id: 0
    rmw_implementation: rmw_fastrtps_cpp
platform:
  controller: ps4
  decorations:
    front_bumper:
      enabled: true
      model: default
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      extension: 0.0
    rear_bumper:
      enabled: true
      model: default
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      extension: 0.0
    structure:
      enabled: true
      model: sensor_arch_300
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
    top_plate:
      enabled: true
      model: pacs
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
  extras:
    urdf: null
    control: null
accessories:
  box:
    - name: user_bay_cover
      parent: top_plate_link
      xyz: [0.0, 0.0, 0.00735]
      rpy: [0.0, 0.0, 0.0]
      size: [0.4, 0.4, 0.002]
  cylinder: []
  link: []
  mesh: []
  sphere: []
mounts:
  bracket: []
  fath_pivot: []
  riser: []
sensors:
  camera: []
  gps: []
  imu: []
  lidar2d: []
  lidar3d: []
```

It is supposed to get this:

[image]

 **Actual behaviour**
 I get this: 

[image]

**To Reproduce**
Simply follow your instructions:
1. Clone the above packages in a workspace and build
2. store the robot.yaml file I shown before in `~/clearpath/`
3. ros2 launch clearpath_gz simulation.launch.py

roni-kreinin, 2023-08-10:
Can you try launching the simulation with 

```
ros2 launch clearpath_gz simulation.launch.py setup_path:=$HOME/clearpath/
```

The default setup path is `/etc/clearpath/`.

fmrico, 2023-08-10:
Hi @roni-kreinin 

I have changed ~/clearpath/robot.yaml as you told me in #12 

[robot.zip](https://github.com/clearpathrobotics/clearpath_simulator/files/12315764/robot.zip)

I Attach the output of:
1. `ros2 launch clearpath_gz gz_sim.launch.py &> gz_out.txt`
2. `ros2 launch clearpath_gz robot_spawn.launch.py &> spawn_out.txt`
3. `ros2 run rviz2 rviz2 --ros-args -p use_sim_time:=True &> rviz2_out.txt`

Now I can see the robot as the next image, but I cannot visualize pointcloud, laser, or image (blue image?) from sensors.
[image]

I hope it helps
[gz_out.txt](https://github.com/clearpathrobotics/clearpath_simulator/files/12315759/gz_out.txt)
[rviz2_out.txt](https://github.com/clearpathrobotics/clearpath_simulator/files/12315760/rviz2_out.txt)
[spawn_out.txt](https://github.com/clearpathrobotics/clearpath_simulator/files/12315761/spawn_out.txt)

tonybaltovski, 2023-08-31:
@fmrico can update to last branches for all?

closed 2023-09-05 by tonybaltovski