clearpathrobotics/clearpath_simulator #13: Jackal does not appear in Gazebo simulation

kvpz, 2023-08-11:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source or installed: installed
 - Package version: 0.0.2-1jammy.20230722.064635 (from repository)
 
 **Expected behaviour**
Expected to see a simulated Jackal within the Gazebo Ignition simulator. 

 **Actual behaviour**
[image]

**To Reproduce**
Provide the steps to reproduce:
1. Run ```ros2 launch clearpath_gz simulation.launch.py```
2. Observe the following lines printed in the terminal:
```
[INFO] [launch]: All log files can be found below /home/username/.ros/log/2023-08-11-15-02-43-473696-userhostname-11059
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): No module named 'clearpath_config.common'
```
3. Observe Gazebo simulator starts

**Other notes**
I ran into this issue while following the Clearpath Robotics tutorial: [Tutorials > Simulator > Simulate](https://docs.clearpathrobotics.com/docs/ros/tutorials/simulator/simulate/#launching-the-simulator)

roni-kreinin, 2023-08-14:
@kvpz Can you update your clearpath config package to the latest version?

```
sudo apt update
sudo apt install ros-humble-clearpath-config
```

kvpz, 2023-08-23:
I updated the ros-humble-clearpath-config package today from version 0.0.4 to 0.0.6. I then ran the ```. install/setup.bash``` command in the clearpath_ws workspace. The jackal appeared in the Gazebo simulation when I ran ```ros2 launch clearpath_gz simulation.launch.py```. Everything is fine now. Thanks!

[image]

closed 2023-08-23 by kvpz