clearpathrobotics/clearpath_simulator #14: Robot (husky) not showing in simulation

y2d2, 2023-08-22:
**Please provide the following information:**
 - OS: Ubuntu 22.04 (in docker)
 - ROS 2 Distro: Humble
 - Built from source or installed: from source
 - Package version: not sure which packages are relevant 
  -- clearpath-common: commit 54287f3ba562b5c353c100e61a4dd8e5963dfdfa 
  -- clearpath-desktop: commit 2d70a3f8dc0c2d3ecef5cc99866e2b6b6c8a6dd6
  -- clearpath-simulator: commit 35078baa08990d4a4212e13095ef24e97daf4472

 **Expected behaviour**
when I run the command:  `ros2 launch clearpath_gz simulation.launch.py` I expect to see the simulator starting with the husky loaded for which the `robot.yaml` has been updated (with the yaml from https://docs.clearpathrobotics.com/docs/ros/config/yaml/ updated with the correction for the hokuyo_ust10 to hokuyo_ust.) 

 **Actual behaviour**
The warehouse loads, however, no robot is visible and the launch file stops. (without closing the gazebo). 
The following output might give more insight: 
```
root@husky:/# ros2 launch clearpath_gz simulation.launch.py
...
[generate_description-3]   File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/utils/dictionary.py", line 34, in _flatten_dict_gen
[generate_description-3]     for k, v in d.items():
[generate_description-3] AttributeError: 'str' object has no attribute 'items'
[ERROR] [generate_description-3]: process has died [pid 262, exit code 1, cmd '/root/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_description -s /root/clearpath/ --ros-args -r __node:=generate_description'].
...
[generate_launch-4]   File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/utils/dictionary.py", line 34, in _flatten_dict_gen
[generate_launch-4]     for k, v in d.items():
[generate_launch-4] AttributeError: 'str' object has no attribute 'items'
[ERROR] [generate_launch-4]: process has died [pid 276, exit code 1, cmd '/root/clearpath_ws/install/clearpath_generator_gz/lib/clearpath_generator_gz/generate_launch -s /root/clearpath/ --ros-args -r __node:=generate_launch'].
...
[generate_param-5]   File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/utils/dictionary.py", line 34, in _flatten_dict_gen
[generate_param-5]     for k, v in d.items():
[generate_param-5] AttributeError: 'str' object has no attribute 'items'
[ERROR] [generate_param-5]: process has died [pid 284, exit code 1, cmd '/root/clearpath_ws/install/clearpath_generator_gz/lib/clearpath_generator_gz/generate_param -s /root/clearpath/ --ros-args -r __node:=generate_launch'].
[ERROR] [launch]: Caught exception in launch (see debug for traceback): [Errno 2] No such file or directory: '/root/clearpath/platform/launch/platform-service.launch.py'
...
[INFO] [parameter_bridge-2]: sending signal 'SIGINT' to process[parameter_bridge-2]
[INFO] [ruby $(which ign) gazebo-1]: sending signal 'SIGINT' to process[ruby $(which ign) gazebo-1]
[parameter_bridge-2] [INFO] [1692707046.603289943] [rclcpp]: signal_handler(signum=2)
[parameter_bridge-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[parameter_bridge-2]   what():  Couldn't initialize rcl timer handle: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67
[ERROR] [parameter_bridge-2]: process has died [pid 259, exit code -6, cmd '/opt/ros/humble/lib/ros_gz_bridge/parameter_bridge /clock@rosgraph_msgs/msg/Clock[ignition.msgs.Clock --ros-args -r __node:=clock_bridge'].
...
```
The full output of the command :
[Simulation_issue.txt](https://github.com/clearpathrobotics/clearpath_simulator/files/12409745/Simulation_issue.txt)

**To Reproduce**
Provide the steps to reproduce:
1.Fresh ubuntu22.04 docker with ros-humble install
2. cloned the latest version of   clearpath-common,  clearpath-desktop, clearpath-simulator and build
3. Create the robot.yaml and set the setup commands. 
4. run `ros2 launch clearpath_gz simulation.launch.py`

**Other notes**
I tried the same from the install repos, but ran into the issue of the hokuyo_ust. But the result is the same the gazebo loads without the robot showing.

hilary-luo, 2023-08-24:
Please see PR https://github.com/clearpathrobotics/clearpath_simulator/pull/15

You can build from source using the indicated branches or wait for these PR's to be merged and the updates released.

y2d2, 2023-08-28:
Ok thanks!

closed 2023-08-28 by y2d2