clearpathrobotics/clearpath_simulator #18: Warehouse Simulation not showing robot model

immanueln98, 2023-09-21:
Good Day

I am having trouble displaying the robot model in the warehouse when running **ros2 launch clearpath_gz simulation.launch.py**.
The robot model does not seem to be displayed and I get the following error in the terminal.
[](url)
[image]
Can I please have some assistance please? I have followed all instructions in setting up the remote computer for simulation.

hilary-luo, 2023-09-21:
Can you please provide the full terminal output and the robot.yaml file that you are using?

kvpz, 2023-09-21:
Hi @immanueln98 I also had the same issue when the jackal was equipped with the hokuyo_ust 2d lidar model. The reason why the jackal was not appearing for me within the simulation was because the data member IP_ADDRESS of the BaseLidar2d class was assigned the incorrect IP address 192.168.131.20. Try changing the ip address to 192.168.131.25. This is the file you need to edit: https://github.com/clearpathrobotics/clearpath_config/blob/main/clearpath_config/sensors/types/lidars_2d.py 

See if that helps. And please share the robot.yaml file as @hilary-luo asked regardless if that solves your issue or not so we can get a better view of your setup

hilary-luo, 2023-09-21:
@kvpz if you are needing to configure the IP address for your hokuyo the intended way is to pass it through using the robot.yaml file, not by changing the source files. It is one of the ros parameters that gets passed through and thus you can also configure multiple lidars with different ip addresses on the same robot. See the documentation [here](https://docs.clearpathrobotics.com/docs/ros/config/yaml#lidar-2d) or the following excerpt from a robot.yaml file:

```
lidar2d:
  - model: hokuyo_ust
    urdf_enabled: true
    launch_enabled: true
    parent: base_link
    xyz: [0.0, 0.0, 0.0]
    rpy: [0.0, 0.0, 0.0]
    ros_parameters:
      urg_node:
        laser_frame_id: lidar2d_0_laser
        ip_address: 192.168.131.20
        ip_port: 10940
        angle_min: -3.141592653589793
        angle_max: 3.141592653589793
````

An example of this already being customized is the [dual laser sample config](https://github.com/clearpathrobotics/clearpath_config/blob/main/clearpath_config/sample/j100/j100_dual_laser.yaml). There you can see two different lidars with 2 different IP addresses. If this is not working for you then we can troubleshoot that particular feature.

immanueln98, 2023-09-22:
Thank you all for your responses. Below is the full terminal output when launching the simulation file:

`[INFO] [launch]: All log files can be found below /home/imanuel/.ros/log/2023-09-22-11-47-21-179211-imanuel-ROG-Zephyrus-G14-GA401IV-GA401IV-4163
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [4164]
[INFO] [parameter_bridge-2]: process started with pid [4166]
[INFO] [generate_description-3]: process started with pid [4169]
[parameter_bridge-2] [INFO] [1695376041.694788877] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[generate_description-3] Traceback (most recent call last):
[generate_description-3]   File "/home/imanuel/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_description", line 46, in <module>
[generate_description-3]     main()
[generate_description-3]   File "/home/imanuel/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_description", line 42, in main
[generate_description-3]     dg.generate()
[generate_description-3]   File "/home/imanuel/clearpath_ws/install/clearpath_generator_common/local/lib/python3.10/dist-packages/clearpath_generator_common/description/generator.py", line 74, in generate
[generate_description-3]     self.generate_sensors()
[generate_description-3]   File "/home/imanuel/clearpath_ws/install/clearpath_generator_common/local/lib/python3.10/dist-packages/clearpath_generator_common/description/generator.py", line 166, in generate_sensors
[generate_description-3]     sensors = self.clearpath_config.sensors.get_all_sensors()
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/clearpath_config.py", line 187, in sensors
[generate_description-3]     self._sensors.config[self.SENSORS])
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/common/types/config.py", line 87, in config
[generate_description-3]     self.getter(prop)()
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/sensors/sensors.py", line 303, in camera
[generate_description-3]     value=self._camera.to_dict()
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/sensors/sensors.py", line 219, in to_dict
[generate_description-3]     d.append(accessory.to_dict())
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/sensors/types/sensor.py", line 97, in to_dict
[generate_description-3]     d['ros_parameters'] = self.get_ros_parameters()
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/sensors/types/sensor.py", line 214, in get_ros_parameters
[generate_description-3]     return self.ros_parameters
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/sensors/types/sensor.py", line 198, in ros_parameters
[generate_description-3]     d[node_name] = flatten_dict(d[node_name])
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/common/utils/dictionary.py", line 43, in flatten_dict
[generate_description-3]     return dict(_flatten_dict_gen(d, parent_key, dlim))
[generate_description-3]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/common/utils/dictionary.py", line 34, in _flatten_dict_gen
[generate_description-3]     for k, v in d.items():
[generate_description-3] AttributeError: 'str' object has no attribute 'items'
[ERROR] [generate_description-3]: process has died [pid 4169, exit code 1, cmd '/home/imanuel/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_description -s /home/imanuel/clearpath/ --ros-args -r __node:=generate_description'].
[INFO] [generate_launch-4]: process started with pid [4187]
[generate_launch-4] Traceback (most recent call last):
[generate_launch-4]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_launch", line 46, in <module>
[generate_launch-4]     main()
[generate_launch-4]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_launch", line 42, in main
[generate_launch-4]     glg.generate()
[generate_launch-4]   File "/home/imanuel/clearpath_ws/install/clearpath_generator_common/local/lib/python3.10/dist-packages/clearpath_generator_common/launch/generator.py", line 79, in generate
[generate_launch-4]     self.generate_sensors()
[generate_launch-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_gz/launch/generator.py", line 208, in generate_sensors
[generate_launch-4]     sensors = self.clearpath_config.sensors.get_all_sensors()
[generate_launch-4]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/clearpath_config.py", line 187, in sensors
[generate_launch-4]     self._sensors.config[self.SENSORS])
[generate_launch-4]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/common/types/config.py", line 87, in config
[generate_launch-4]     self.getter(prop)()
[generate_launch-4]   File "/home/imanuel/clearpath_ws/build/clearpath_config/clearpath_config/sensors/sensors.py", line 303, in camera
[generate_launch-4]     value=self._camera.to_dict()
[... cut ...]
        enable_depth: true
        depth_module.profile: 640,480,30
        pointcloud.enable: true
  gps: []
  imu: []
  lidar3d:
    - model: velodyne_lidar
      urdf_enabled: true
      launch_enabled: true
      parent: sensor_arch_mount
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      ros_parameters:
        frame_id: lidar3d_0_laser
        device_ip: 192.168.131.25
        port: 2368
        model: VLP16
        fixed_frame: lidar3d_0_laser
        target_frame: lidar3d_0_laser`

hilary-luo, 2023-09-25:
If you look at the terminal output, you will see that the first error message you get is the following:
```
[generate_description-3] AttributeError: 'str' object has no attribute 'items'
[ERROR] [generate_description-3]: process has died [pid 4169, exit code 1, cmd '/home/imanuel/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_description -s /home/imanuel/clearpath/ --ros-args -r __node:=generate_description'].
```

This error is indicating that it was not able to fully generate the robot description files from the robot.yaml and therefore that there is a problem with the robot.yaml file. This is a lot easier to see if you view your model first and ensure that the robot.yaml is good before using it in a simulation. 

There are a few different issues with your robot.yaml. Take a look at the following compared to the one you provided:

```
serial_number: a200-0656
version: 0
system:
  hosts:
    self: cpr-a200-0656
    platform:
      cpr-a200-0656: 192.168.131.1
    onboard: {}
    remote: {}
  ros2:
    username: administrator
    namespace: a200_0656
    domain_id: 0
    rmw_implementation: rmw_fastrtps_cpp
    workspaces:
    - /home/imanuel/clearpath_ws/install/setup.bash
platform:
  controller: ps4
  attachments:
    front_bumper:
      enabled: true
      model: default
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      extension: 0.0
    rear_bumper:
      enabled: true
      model: default
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      extension: 0.0
    structure:
      enabled: true
      model: sensor_arch_300
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
    top_plate:
      enabled: true
      model: pacs
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
  extras:
    urdf: /home/imanuel/clearpath_ws/src/clearpath_common/clearpath_platform_description/urdf/a200/a200.urdf.xacro
    ros_parameters:
      platform_velocity_controller:
        wheel_radius": 0.1651
        linear.x.max_velocity": 1.0
        linear.x.min_velocity": -1.0
        linear.x.max_acceleration": 3.0
        linear.x.min_acceleration": -3.0
        angular.z.max_velocity": 2.0
        angular.z.min_velocity": -2.0
        angular.z.max_acceleration": 6.0
        angular.z.min_acceleration": -6.0
links:
  box:
    - name: user_bay_cover
      parent: top_plate_link
      xyz: [0.0, 0.0, 0.00735]
      rpy: [0.0, 0.0, 0.0]
      size: [0.4, 0.4, 0.002]
  cylinder: []
  link: []
  mesh: []
  sphere: []
mounts:
  bracket:
    - parent: top_plate_mount_d1
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      model: horizontal
  fath_pivot:
    - parent: sensor_arch_mount
      xyz: [0.0, 0.0, -0.021]
      rpy: [3.1415, 0.0, 0.0]
      angle: 0.0
  riser: []
sensors:
  camera:
    - model: intel_realsense
      urdf_enabled: true
      launch_enabled: true
      parent: fath_pivot_0_mount
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      ros_parameters:
        camera:
          camera_name: camera_0
          device_type: d435
          serial_no: "0"
          enable_color: true
          rgb_camera.profile: 640,480,30
          enable_depth: true
          depth_module.profile: 640,480,30
          pointcloud.enable: true
  gps: []
  imu: []
  lidar3d:
    - model: velodyne_lidar
      urdf_enabled: true
      launch_enabled: true
      parent: sensor_arch_mount
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      ros_parameters:
        velodyne_driver_node:
          frame_id: lidar3d_0_laser
          device_ip: 192.168.131.25
          port: 2368
          model: VLP16
        velodyne_convert_node:
          model: VLP16
          fixed_frame: lidar3d_0_laser
          target_frame: lidar3d_0_laser
```

I wasn't able to fully test this out because I don't have your extras urdf, so see how this works for you and what error messages you get. Feel free to respond with your results (and if you would like me to simulate it then also all additional required files). I would suggest copying the robot.yaml into `/etc/clearpath/` on your computer and then using the [live yaml](https://docs.clearpathrobotics.com/docs/ros/config/live) functionality to verify and modify your model (soon you should be able to put it in any folder but for now you have to place it there to use this tool and make sure you have permissions set up to edit it). While using this, watch the terminal output for errors or crashing. If it has crashed then you will need to make the appropriate change in your robot.yaml and then relaunch the tool. 

Do you mind me asking how you came up with your robot.yaml? are there any improvements that you would suggest to the documentation or sample codes that would have helped you build it correctly? A better method to modify this file and check it for errors is planned but unscheduled currently so the focus for this question is on documentation.

immanueln98, 2023-09-25:
Thank you for your response. I will make the appropriate changes. With regards to the added urdf, I was not sure which specific urdf should go there or if it is necessary for the bare bones a200 robot model. So i just pointed to the urdf provided within the clearpath_common package folder.

As for how I came up with this model, I used the provided sample robot.yaml sections provided in the documentation. The installation section of the documentation is good and understandable. However, the tutorial navigation section of the docs does not touch on how the demo's can be run on the physical robot, it just states that it will be done.

immanueln98, 2023-09-25:
[image]
I tried running the view_model launch script in order to use the live yaml functionality but I am getting the above error and I can't seem to figure out how to fix it.

hilary-luo, 2023-09-28:
>Thank you for your response. I will make the appropriate changes. With regards to the added urdf, I was not sure which specific urdf should go there or if it is necessary for the bare bones a200 robot model. So i just pointed to the urdf provided within the clearpath_common package folder.

The urdf section is only if you are wanting to add additional components on top of the base components / the components that have been added as part of the robot.yaml file - that is why it is under the 'extras' section. So if you are not adding anything additional then feel free to set the extras urdf back to null. You do not need the clearpath packages cloned into your workspace either if you are not making changes to anything.

> As for how I came up with this model, I used the provided sample robot.yaml sections provided in the documentation. The installation section of the documentation is good and understandable. However, [...]

It would be worth reviewing the documentation again then because some of the things you had in your robot.yaml were very outdated, from the early days of the development of this project. We don't intend to be making breaking changes to the robot.yaml in the long term but it has been necessary during the initial development prior to the official release packages. In regards to running navigation, the only difference should be to skip the simulation step. You can directly launch rviz and then slam or localization or nav2 (depending on the tutorial that you are following). Just make sure to have the robot.yaml at the location you indicate in the setup_path parameter.

>I tried running the view_model launch script in order to use the live yaml functionality but I am getting the above error and I can't seem to figure out how to fix it. Not sure where or how to include the tag files name.

That error is about it not being able to find the extras urdf that you linked. I assume that you are doing this on your pc and that file is on the robot. Either way it is a moot point as we have now discussed that that xacro should not be included in the robot.yaml

>Also, when I moved my .yaml file to /etc/clearpath as you suggested, gazebo can't find the yaml file anymore when running the simulation script as shown below.

We did just discover that there is an issue with that and the fix will be released shortly and the documentation will be updated to explain how to indicate where the robot.yaml has been placed. As I am guessing you figured out already, for now continue with placing this file in the /etc/clearpath folder on your computer until the fix is released.

immanueln98, 2023-10-02:
Thanks Hilary for your reponses and clarifications! I will await your fix. Any idea on when that will likely happen?

Also one quick question, when running these nav demos on the physical robot, all I need to do when launching them is adding the argument "use_sim_time:=false" right?

immanueln98, 2023-10-05:
I managed to get the Husky to show up in Gazebo, however it does not move when I input velocity commands.
[image]

Rviz also has trouble displaying the robot model from the robot_description topic. Nothing seems to be published from it.
[image]

hilary-luo, 2023-10-06:
> Thanks Hilary for your reponses and clarifications! I will await your fix. Any idea on when that will likely happen?

The fix should be out now, the documentation should be updated 

>Also one quick question, when running these nav demos on the physical robot, all I need to do when launching them is adding the argument "use_sim_time:=false" right?

You can to be certain, but the default should be false so you shouldn't need to explicitly state this. 

>I managed to get the Husky to show up in Gazebo, however it does not move when I input velocity commands.

It doesn't look like you have pressed the play button to start the simulation: 
[image]
or added the namespacing for the cmdvel topic:
[image]

immanueln98, 2023-10-09:
Hi Hilary

Thanks for your help! I got it moving in gazebo now!

closed 2023-10-09 by immanueln98