clearpathrobotics/clearpath_simulator #29: [ERROR] yaml file not found

Peenut173, 2024-01-22:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS 2 Distro: (e.g. Humble)

 **Expected behaviour**
I clone the git repository, built all the packages (which went well) and then tried to run the simulation launch file from clearpath_gz folder. Then it should start the simulation with a robot.

 **Actual behaviour**
Actually, Gazebo runs well, I have the window with everything in it but the robot and I get this message in my terminal:
[ERROR] [launch]: Caught exception in launch (see debug for traceback): YAML file 'None' could not be found

I suppose it is related to the launch file not finding the robot.yaml file but I'm not quite sure about that.

**To Reproduce**
Provide the steps to reproduce:
1. clone the repository
2. colon build --symlink-install
3. ros2 launch learpath_gz simulation.launch.py

tonybaltovski, 2024-01-22:
Did you create a `robot.yaml`?: https://docs.clearpathrobotics.com/docs/ros/config/yaml/system

Peenut173, 2024-01-22:
No, but I have a husky.yaml file with the same kind of implementation in a clearpath folder. I was thinking of replacing robot.yaml by husky.yaml in my launch file but I don't know if it is correct

tonybaltovski, 2024-01-22:
The generators look for `robot.yaml` to generate the needed launch files.  Could you copy [this](https://github.com/clearpathrobotics/clearpath_config/blob/main/clearpath_config/sample/a200/a200_dual_laser.yaml) as your `robot.yaml` and try again?

Peenut173, 2024-01-22:
I can try, where should I create the yaml file?

Peenut173, 2024-01-22:
okay, i created the robot file in the same folder as my previous husky one. I get another error message: [ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /home/peenut/clearpath/robot.urdf.xacro is_sim:=true gazebo_controllers:=/home/peenut/clearpath/platform/config/control.yaml namespace:=a200_0000
Captured stderr output: error: No such file or directory: /.urdf.xacro [Errno 2] No such file or directory: '/.urdf.xacro'
when processing file: /home/peenut/clearpath/robot.urdf.xacro

and:

[ERROR] [ruby $(which ign) gazebo-1]: process[ruby $(which ign) gazebo-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'

[ERROR] [ruby $(which ign) gazebo-1]: process has died [pid 7390, exit code -15, cmd 'ruby $(which ign) gazebo warehouse.sdf -v 4 --gui-config /media/peenut/programmes/ros_projects/cs8813/husky/clearpath_ws/install/clearpath_gz/share/clearpath_gz/config/gui.config --force-version 6'].

Peenut173, 2024-01-23:
certainly, here is the file: serial_number: a200-0000
version: 0
system:
  username: administrator
  hosts:
    self: cpr-a200-0000
    platform:
      cpr-a200-0000: 192.168.131.1
    onboard: {}
    remote: {}
  ros2:
    namespace: a200_0000
    domain_id: 0
    rmw_implementation: rmw_fastrtps_cpp
    workspaces: []
platform:
  controller: ps4
  battery:
    model: ES20_12C
    configuration: S2P1
  attachments:
    - name: front_bumper
      type: bumper
      parent: front_bumper_mount
    - name: rear_bumper
      type: bumper
      parent: rear_bumper_mount
    - name: top_plate
      type: top_plate
    - name: sensor_arch
      type: sensor_arch
      parent: default_mount
      enabled: False
  extras:
    urdf: null
    ros_parameters:
      platform_velocity_controller:
        linear.x.max_velocity": 1.0
        linear.x.min_velocity": -1.0
        linear.x.max_acceleration": 3.0
        linear.x.min_acceleration": -3.0
        angular.z.max_velocity": 2.0
        angular.z.min_velocity": -2.0
        angular.z.max_acceleration": 6.0
        angular.z.min_acceleration": -6.0
links:
  box: []
  cylinder: []
  frame: []
  mesh: []
  sphere: []
mounts:
  bracket: []
  fath_pivot: []
  riser: []
  disk: []
  post: []
sensors:
  camera: []
  gps: []
  imu: []
  lidar2d: []
  lidar3d: []

Peenut173, 2024-01-23:
I did it, but still have the error messages

luis-camero, 2024-01-23:
Hi @Peenut173, would you mind checking that you have the latest changes, i.e. `git pull` in `clearpath_config`. 

Also, please attach the generated `/home/peenut/clearpath/robot.urdf.xacro` so we can take a look at where it is that the `'/.urdf.xacro'` is being referenced.

Peenut173, 2024-01-23:
I just saw that robot.urdf.xacro is in /media/peenut/programmes/ros_projects/cs8813/husky/clearpath_ws/src/clearpath/. husky is the folder where we clone the repository. Here is what the file contains:

<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="a200-0000">

  
  <xacro:include filename="$(find clearpath_platform_description)/urdf/common.urdf.xacro"/>

  
  <xacro:include filename="$(find clearpath_platform_description)/urdf/a200/a200.urdf.xacro"/>
  <xacro:a200 wheel="default"/>

  

  <xacro:include filename="$(find clearpath_platform_description)/urdf/a200/attachments/bumper.urdf.xacro"/>
  <xacro:bumper name="front_bumper" model="default" parent_link="front_bumper_mount" extension="0.0">
    <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
  </xacro:bumper>

  <xacro:include filename="$(find clearpath_platform_description)/urdf/a200/attachments/bumper.urdf.xacro"/>
  <xacro:bumper name="rear_bumper" model="default" parent_link="rear_bumper_mount" extension="0.0">
    <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
  </xacro:bumper>

  <xacro:include filename="$(find clearpath_platform_description)/urdf/a200/attachments/top_plate.urdf.xacro"/>
  <xacro:top_plate name="top_plate" model="default" parent_link="default_mount">
    <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
  </xacro:top_plate>

  

  

  

  

  <xacro:include filename="/.urdf.xacro"/>

</robot>

Also I have no clearpath folder in /home/peenut, which is also something I have just noticed. I finally can not find any clearpath_config in my folder. I believe this is not normal but we just cloned the git in husky folder, so it should be there if I am not mistaken?

luis-camero, 2024-01-23:
The `clearpath_config` should be under `/media/peenut/programmes/ros_projects/cs8813/husky/clearpath_ws/src` where you seem to have cloned and built it. That is where you should call `git pull` to get the latest changes.

Peenut173, 2024-01-23:
Okay, I got the clearpath_config repository and now I see the robot on the simulation. I still have some error messages, for instance:
[ERROR] [1706049581.601866904] [a200_0000.joy_node]: Couldn't open joystick /dev/input/ps4. Will retry every second.

But I think this is due to something else, like a node that is not running and as soon as I'll run it the error message will disappear if I am correct?

luis-camero, 2024-01-24:
Yeah, that is expected behaviour if you do not have a PS4 controller connected. It is the standard we use for our robots, so we use it on simulations too. 

It seems that your issue is now resolved and you no longer have an issue launching the simulation.

closed 2024-01-24 by luis-camero