clearpathrobotics/clearpath_simulator #30: Warehouse world is broken

pfranklincra, 2024-03-21:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source or installed: Both
 - Package version: 874993b370c99d109775a706a7566884048264c4

 **Expected behaviour**
Gazebo world with warehouse launches, as described in tutorials

 **Actual behaviour**
Gazebo fails to launch, throws errors indicating an attempt to download from fuel.ignitionrobotics.org which is deprecated.

**To Reproduce**
Provide the steps to reproduce:
1. Follow instructions to install clearpath_simulator on a fresh machine (or docker image)
2. ros2 launch clearpath_gz simulation.launch.py
3. see the error

**Other notes**

The current fix I'm using is commenting out the MovAI mobile pallet and Rescue Randy from the warehouse.sdf file. These both appear to reference the old fuel.ignitionrobotics.org url, which is now unavailable.

Relevant console output from Gazebo (truncated):

```
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org

```

```
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: openrobotics/models/artifact proximity detector/tip/artifact proximity detector.zip
[ruby $(which ign) gazebo-1]   REST response code: 504
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/model[@name="rescue_randy_sitting"]/include[0]/uri:/root/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/rescue randy sitting/8/model.sdf:L6]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Artifact Proximity Detector]

```

hilary-luo, 2024-03-26:
I have been able to replicate this issue. In the meantime, those two models can be commented out from source installs and don't use the depot world, or you can download and place each file manually in your .ignition folder.

For example:
Download from this link [https://fuel.gazebosim.org/1.0/OpenRobotics/models/artifact%20proximity%20detector.zip](https://fuel.gazebosim.org/1.0/OpenRobotics/models/artifact%20proximity%20detector.zip) and unzip and place the two files it contains at `$HOME/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/ artifact proximity detector/6/`

The old url is still mostly working (despite deprecation). It is the .zip links embedded in those open robotics models that are causing the errors.

hilary-luo, 2024-03-27:
@pfranklincra Can you confirm if this issue has been resolved for you now?

hilary-luo, 2024-03-27:
It seems to have issues again this morning so if you have not tested it yet, it likely won't work currently.

pfranklincra, 2024-03-28:
ah darn, I just tested it now before seeing your most recent message. I'm still getting a 504 error, guess I missed the window for it to be working :P

Regardless the workarounds do work, so this is not a huge concern for us.

ajjaya18, 2024-04-21:
> I have been able to replicate this issue. In the meantime, those two models can be commented out from source installs, or you can download and place each file manually in your .ignition folder.
> 
> For example: Download from this link https://fuel.gazebosim.org/1.0/OpenRobotics/models/artifact%20proximity%20detector.zip and unzip and place the two files it contains at `$HOME/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/ artifact proximity detector/6/`
> 
> The old url is still mostly working (despite deprecation). It is the .zip links embedded in those open robotics models that are causing the errors.

I try this solution but the files in this zip is the same that I already have it. 
[image]

hilary-luo, 2024-04-30:
@ajjaya18 That is an example link and you need to execute that process for the files that are specifically giving you errors. The errors would be in regards to files that you do not have.

closed 2024-07-24 by tonybaltovski