clearpathrobotics/clearpath_simulator #44: imu_0_gz_bridge bridges topic sensors/imu_0/data which is not produced by Gazebo

AdamGoertz, 2024-08-22:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source
 - Package version: 9dd763df8293da5008aeb43979b4dfd5592f0e4d

 **Expected behaviour**
 GZ_TO_ROS bridges should only be created for topics that are actually published by gazebo.

 **Actual behaviour**
 `imu_0_gz_bridge` creates a publisher for the `sensors/imu_0/data` topic even though that topic is published by the `imu_filter_node`.

**To Reproduce**
Provide the steps to reproduce:
1. Run the simulator `ros2 launch clearpath_gz simulation.launch.py`
2. `ros2 topic info /w200_0000/sensors/imu_0/data -v`
3. Output:

```
Type: sensor_msgs/msg/Imu

Publisher count: 2

Node name: imu_0_gz_bridge
Node namespace: /w200_0000
Topic type: sensor_msgs/msg/Imu
Endpoint type: PUBLISHER
GID: 01.0f.73.dd.97.b8.a3.ad.00.00.00.00.00.00.13.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: imu_filter_node
Node namespace: /w200_0000
Topic type: sensor_msgs/msg/Imu
Endpoint type: PUBLISHER
GID: 01.0f.73.dd.99.b8.25.22.00.00.00.00.00.00.21.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: ekf_node
Node namespace: /w200_0000
Topic type: sensor_msgs/msg/Imu
Endpoint type: SUBSCRIPTION
GID: 01.0f.73.dd.89.b8.ca.2a.00.00.00.00.00.00.21.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

This issue can be fixed by deleting these lines

https://github.com/clearpathrobotics/clearpath_simulator/blob/9dd763df8293da5008aeb43979b4dfd5592f0e4d/clearpath_generator_gz/clearpath_generator_gz/param/sensors.py#L196-L201

Is there a configuration where it is expected that the simulator will publish the filtered IMU topic? Or where `sensors/imu_0/data` actually refers to the raw IMU data (instead of the filtered output)? A quick look suggests all of the URDFs in clearpath_platform_description use the `data_raw` topic.

luis-camero, 2024-08-23:
I understand the confusion here. As you have correctly pointed out, on all platforms that have built-in IMU's, the data coming in from the MCU must be filtered. Therefore, for platform IMU's we only need to bridge the `data_raw` topic to then filter it. 

On the other hand, we support external IMU's that can be added to improve robot localization. These IMU's, like the [`microstrain_imu`](https://github.com/clearpathrobotics/clearpath_common/blob/71028f322050d6dcadc5b0046631ef2c149a0d66/clearpath_sensors_description/urdf/microstrain_imu.urdf.xacro#L33), have independent drivers that allow for calibration and data filtering, resulting in `data` topics that are assumed to be much more accurate. 

In order to support both with just a single `IMU` class of sensors, we opted to bridge both topics, assuming that one of the two would be used. It is not desirable and we will be making an update to prevent unnecessary bridges from being generated.

closed 2024-08-23 by luis-camero

AdamGoertz, 2024-08-23:
Got it, thanks for the explanation!