clearpathrobotics/clearpath_simulator #46: Switch Controller fails launch after 5 seconds, unable to control Husky in simulation

HUEY-9384, 2024-09-05:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source or installed: Installed
 - Package version: 0.2.6-1jammy.20240823.154258 and 0.2.6-1jammy.20240813.035058

 **Expected behaviour**
lunched properly and it able to move with controller or keyboard or gazebo control panel

 **Actual behaviour**
 [ruby $(which ign) gazebo-1] [WARN] [1725572690.821943106] [a200_0000.platform_velocity_controller]: Can't accept new commands. subscriber is inactive
and it wont move

**To Reproduce**
Provide the steps to reproduce:
1. followed all instruction that provided in https://docs.clearpathrobotics.com/docs/ros/
2.  run simulation

https://pastebin.com/V1ME57he

**Other notes**
its sometimes randomly works fine but mostly not working

luis-camero, 2024-09-06:
We have also noticed this issue. Although we are still searching for the reason and solution to the issue, it seems that the problem is caused by a timeout in the controller manager's state machine, and that if the simulation is started (i.e. the 'Play' button is pressed) within 5 seconds of the instantiation of the controllers, then this issue does not occur. 

Please, let us know if this temporary workaround is successful for you, and wen will update this ticket when we find a fix.

AltinayGrass, 2024-09-08:
what do you think about add -r option to gazebo start command? 

```    # Gazebo Simulator
    gz_sim = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([gz_sim_launch]),
        launch_arguments=[
            ('gz_args', [LaunchConfiguration('world'),
                         '.sdf',
                         ' -v 4',
                         ' -r',
                         ' --gui-config ',
                         gui_config])
        ]
    )
```

Jeehoon-K, 2024-10-07:
I'm having the same issue. It does work if I press 'Play' within 5 sec. Has anyone found a solution to this?

closed 2024-11-04 by hilary-luo