clearpathrobotics/clearpath_simulator #50: Simulation cannot activate controller

ZhongmouLi, 2024-10-15:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Humble
 -  installed:
 - Package version: 0.2.6-1jammy.20240911.170638

 **Expected behaviour**
lunched simulation properly and it able to move with keyboard controller or gazebo control panel

 **Actual behaviour**
[spawner-8] [ERROR] [1728999008.200920895] [a200_0000.spawner_platform_velocity_controller]: Failed to activate controller

[image]

**To Reproduce**
Provide the steps to reproduce:
1. install simulator following https://docs.clearpathrobotics.com/docs/ros/tutorials/simulator/install
2. ros2 launch clearpath_gz simulation.launch.py

**Other notes**
https://pastebin.com/kX8S79Cw

daylanc, 2024-10-28:
@ZhongmouLi, were you able to find a solution? I'm running into the same issue. When attempting to move, I see "Failed to activate controller" and "platform_velocity_controller can't accept new commands. subscriber is inactive."

KushnirDmytro, 2024-10-28:
I've got this one.
Issue is that simulation has to be running

The launch files pipeline is broken and does not allow you to have running sim from the launch file parameters.
You can click the "play" button while simulation is loading, if you have a lucky moment.

Go to pkg `clearpath_gz/launch/gz_sim_launch.gz`

And there when arg `gz_sim = ...` is created adjust to have
```
    # Gazebo Simulator
    gz_sim = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([gz_sim_launch]),
        launch_arguments=[
            ('gz_args', [LaunchConfiguration('world'),
                         '.sdf',
                         ' -v 4 -r ', # Add the "-r" flag to run simulation from start
                         ' --gui-config ',
                         gui_config])
        ]
    )
```
 The elegant solution would be to fix the parameter passing to via this launcher link, but this should fix the issue

ZhongmouLi, 2024-10-28:
@daylanc The solution is what @KushnirDmytro said.

You have to click "Play" as soon as you can. Then it should work.

daylanc, 2024-10-29:
Thank you @KushnirDmytro and @ZhongmouLi. Running the sim on start fixed the problem and I was able to move the rover around!

hilary-luo, 2024-11-04:
Closed as part of https://github.com/clearpathrobotics/clearpath_simulator/pull/52

closed 2024-11-04 by hilary-luo