clearpathrobotics/clearpath_simulator #53: Any plans on upgrading `realsense_gazebo_plugin` and `realsense2_description` packages for the new Gazebo (Gazebo Harmonic or later)

Mechazo11, 2024-10-31:
Hello,

The clearpath_simulator README states compatibility with ROS 2 Humble (LTS) and Ignition Fortress (LTS), with a jazzy branch for ROS 2 Jazzy (LTS). However, [Gazebo's latest docs](https://gazebosim.org/docs/latest/ros_installation/) recommend ROS 2 Jazzy with Gazebo Harmonic (LTS) or newer (e.g., Ionic).

I’ve forward-ported most of ```clearpath_simulator``` to Gazebo Harmonic, testing with the Husky A200 sample configuration from Clearpath's [Robot YAML overview](https://docs.clearpathrobotics.com/docs/ros/config/yaml/overview/). The Husky robot requires ```realsense_gazebo_plugin``` and ```realsense2_description``` packages for the RGBD sensor:

```YAML
sensors:
  camera:
    - model: intel_realsense
      ...
```

These packages, likely from the Gazebo Classic era, may not work beyond Ignition Fortress. My attempt to update ```realsense_gazebo_plugin``` here: https://github.com/Mechazo11/realsense_gazebo_plugin  was not very successful at the time of opening this Issue. I was wondering if Clearpath Robotics had any plans for upgrading these packages for use with Gazebo Harmonic.

With best regards,
@Mechazo11

roni-kreinin, 2024-11-01:
@Mechazo11 We are currently working on upgrading to Jazzy and Harmonic. The Jazzy branch of this repo should work but it is not yet complete and we have not released any Jazzy packages yet. You can check our [docs](https://docs.clearpathrobotics.com/docs/ros/) for updates on Jazzy support.

tonybaltovski, 2025-05-26:
Jazzy released.

closed 2025-05-26 by tonybaltovski