clearpathrobotics/clearpath_simulator #66: Unable to move the robot

HuanyuL, 2024-12-09:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source or installed: Installed
 - Package version: Version: 1.0.0-1jammy.20241128.090538

 **Expected behaviour**
Robot being controlled by the telop panel in gazebo

 **Actual behaviour**
The robot can be initiated but can't move

**To Reproduce**
Provide the steps to reproduce:
1. Launch the simulator
```
ros2 launch clearpath_gz simulation.launch.py setup_path:=$HOME/clearpath
```
3. Edit the topic name according to the namespace in robot yaml.
```
serial_number: a200-0000
version: 0
system:
  hosts:
    - hostname: cpr-a200-0000
      ip: 192.168.131.1
  ros2:
    namespace: a200_0000
platform:
  attachments:
    - name: front_bumper
      type: a200.bumper
      parent: front_bumper_mount
    - name: rear_bumper
      type: a200.bumper
      parent: rear_bumper_mount
    - name: top_plate
      type: a200.top_plate
manipulators:
  arms:
    - model: kinova_gen3_lite
      parent: top_plate_default_mount
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      ip: 192.168.131.40
      port: 10000
      gripper:
        model: kinova_2f_lite
```
5. Recevied the twist msg from ros2 topic echo
6. Errors when launch the simulation
```
[generate_launch-5] Generated launch file: /home/huanyu/clearpath/manipulators/launch/manipulators-service.launch.py
[generate_launch-5] Generated launch file: /home/huanyu/clearpath/sensors/launch/sensors-service.launch.py
[generate_launch-5] Generated launch file: /home/huanyu/clearpath/platform/launch/platform-service.launch.py
[INFO] [generate_launch-5]: process has finished cleanly [pid 380]
[INFO] [generate_param-6]: process started with pid [382]
[generate_param-6] Generated config: /home/huanyu/clearpath/platform/config/control.yaml
[generate_param-6] Traceback (most recent call last):
[generate_param-6]   File "/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py", line 58, in get_package_prefix
[generate_param-6]     content, package_prefix = get_resource('packages', package_name)
[generate_param-6]   File "/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/resources.py", line 86, in get_resource
[generate_param-6]     raise LookupError(
[generate_param-6] LookupError: Could not find the resource 'clearpath_sensors' of type 'packages'
[generate_param-6] 
[generate_param-6] During handling of the above exception, another exception occurred:
[generate_param-6] 
[generate_param-6] Traceback (most recent call last):
[generate_param-6]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_param", line 46, in <module>
[generate_param-6]     main()
[generate_param-6]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_param", line 42, in main
[generate_param-6]     gpg.generate()
[generate_param-6]   File "/dev_ws/install/local/lib/python3.10/dist-packages/clearpath_generator_common/param/generator.py", line 70, in generate
[generate_param-6]     self.generate_platform()
[generate_param-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_gz/param/generator.py", line 70, in generate_platform
[generate_param-6]     platform_param.generate_parameters(use_sim_time=True)
[generate_param-6]   File "/dev_ws/install/local/lib/python3.10/dist-packages/clearpath_generator_common/param/platform.py", line 107, in generate_parameters
[generate_param-6]     self.default_param_file.read()
[generate_param-6]   File "/dev_ws/install/local/lib/python3.10/dist-packages/clearpath_generator_common/common.py", line 193, in read
[generate_param-6]     file_contents = read_yaml(self.full_path)
[generate_param-6]   File "/dev_ws/install/local/lib/python3.10/dist-packages/clearpath_generator_common/common.py", line 176, in full_path
[generate_param-6]     get_package_share_directory(self.package.name), self.path, self.file)
[generate_param-6]   File "/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py", line 80, in get_package_share_directory
[generate_param-6]     path = os.path.join(get_package_prefix(package_name), 'share', package_name)
[generate_param-6]   File "/opt/ros/humble/lib/python3.10/site-packages/ament_index_python/packages.py", line 60, in get_package_prefix
[generate_param-6]     raise PackageNotFoundError(
[generate_param-6] ament_index_python.packages.PackageNotFoundError: "package 'clearpath_sensors' not found, searching: ['/dev_ws/install', '/opt/ros/humble']"
[ERROR] [generate_param-6]: process has died [pid 382, exit code 1, cmd '/opt/ros/humble/lib/clearpath_generator_gz/generate_param -s /home/huanyu/clearpath --ros-args -r __node:=generate_launch'].
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/huanyu/clearpath/platform/config/localization.yaml
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/huanyu/clearpath/platform/config/teleop_interactive_markers.yaml
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/huanyu/clearpath/platform/config/twist_mux.yaml
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/huanyu/clearpath/platform/config/teleop_joy.yaml
```

**Other notes**
Tried both build from source and install but receive the same errors. The simulation works when I switch the ``clearpath_common`` and ``clearpath_config`` back to commit 0.3.4

civerachb-cpr, 2024-12-09:
Thanks for letting us know! It looks like `clearpath_generator_common` is missing a dependency on `clearpath_sensors`, which is included in the `clearpath_robot` repository.

You should be able to manually install the missing dependency by either running `sudo apt-get install ros-humble-clearpath-sensors` or cloning https://github.com/clearpathrobotics/clearpath_robot into your workspace.

PHShaw, 2024-12-09:
Having the same issue, the packages ros-humble-clearpath-sensors or ros-humble-clearpath-robot don't seem to exist?  Cloning and building the clearpath_sensors repository did work though, thanks.

civerachb-cpr, 2024-12-09:
`clearpath_sensors` isn't (currently) available through OSRF's package servers, but it is available through Clearpath Robotocs: https://packages.clearpathrobotics.com/stable/ubuntu/pool/main/r/ros-humble-clearpath-sensors/

See http://packages.clearpathrobotics.com for instructions on how to add Clearpath's package servers to your apt sources.

That said, after looking into this issue more, we're likely going to be doing some restructuring to move the dependencies out of `clearpath_senors` and into a package inside the `clearpath_common` project.  `clearpath_sensors` is intended to provide the launch files for physical sensors, and therefore depends on a lot of additional driver packages you don't necessarily want/need if all you're doing is using the simulator.

I'll update this issue once the fix is officially implemented & released.