clearpathrobotics/clearpath_simulator #69: Using Jackal (j100) imu_0 and gps_0 bridge configuration are not generated

marco-ambrosio, 2025-02-07:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source or installed: installed
 - Package version: 1.0.0-1jammy.20250117.173723

 **Expected behaviour**
When starting a simulation with Jackal [default configuration](https://github.com/clearpathrobotics/clearpath_config/blob/jazzy/clearpath_config/sample/j100/j100_default.yaml) bridges should be created for:
- imu_0
- gps_0

 **Actual behaviour**
Bridges are not created because configurations file `imu_0.yaml` and `gps_0.yaml` are not created in the `sensors/config` folder.

**To Reproduce**
Provide the steps to reproduce:
1. Create a configuration folder and copy the content of [this](https://github.com/clearpathrobotics/clearpath_config/blob/jazzy/clearpath_config/sample/j100/j100_default.yaml) to `robot.yaml`
2. Start the simulation with the command `ros2 launch clearpath_gz simulation.launch.py setup_path:=/path/to/setup/folder`
3. Look for the following errors on the terminal:
```
[parameter_bridge-19] [ERROR] [1738925558.315456455] [readFromYaml]: Could not parse config, top level must be a YAML sequence
[parameter_bridge-17] [ERROR] [1738925559.302815215] [readFromYaml]: Could not parse config, top level must be a YAML sequence
```

**Other notes**
These are the nodes in platform-service.launch.py that generates the error

```python
    node_imu_0_gz_bridge = Node(
        name='imu_0_gz_bridge',
        executable='parameter_bridge',
        package='ros_gz_bridge',
        namespace='j100_0000',
        output='screen',
        parameters=
            [
                {
                    'use_sim_time': True
                    ,
                    'config_file': '/workspaces/marco-humble/misc/clearpath/sensors/config/imu_0.yaml'
                    ,
                }
                ,
            ]
        ,
    )

    node_gps_0_gz_bridge = Node(
        name='gps_0_gz_bridge',
        executable='parameter_bridge',
        package='ros_gz_bridge',
        namespace='j100_0000',
        output='screen',
        parameters=
            [
                {
                    'use_sim_time': True
                    ,
                    'config_file': '/workspaces/marco-humble/misc/clearpath/sensors/config/gps_0.yaml'
                    ,
                }
                ,
            ]
        ,
    )
```

astomodynamics, 2025-02-07:
I have the same issue.

luis-camero, 2025-02-07:
Can you try to pass an absolute path as an argument to the launch file? 
```
ros2 launch clearpath_gz simulation.launch.py setup_path:=/absolute/path/to/setup/folder
```

marco-ambrosio, 2025-02-07:
I solved building from source the packages in `[`clearpath_robot`](https://github.com/clearpathrobotics/clearpath_robot/tree/humble).` Apparently there is a dependency on `clearpath_sensors` package that is not listed in the `package.xml`. 
Moreover, at the time I am writing, I am unable to install the `clearpath_sensors` package from binaries.

luis-camero, 2025-02-10:
We have some dependencies for hardware drivers that prevent us from releasing the `clearpath_robot` packages using the ROS build farm. Instead, we host these builds ourselves. You can find these packages by adding the Clearpath package to your source list: https://wiki.ros.org/ClearpathRobotics/Packages

closed 2025-03-18 by tonybaltovski