clearpathrobotics/clearpath_simulator #83: Publishing to /a300_0000/cmd_vel Not Working

daylanc, 2025-04-25:
**Please provide the following information:**
 - OS: Ubuntu 24.04
 - ROS 2 Distro: Jazzy
 - Built from source or installed: Installed via clearpath_simulator repository
 - Package version: 2.2.0-1noble.20250406.135134

 **Expected behaviour**
 I'm using this [robot.yaml](clearpath_config/sample/a300/a300_default.yaml) file. The simulation and a300 rover spawns fine, but when trying to control the rover from the gazebo sim, it doesn't move. I change the "Topic" to `/a300_0000/cmd_vel` and select "KEYBOARD". I can see data being published when echoing /a300_0000/cmd_vel, but no movement. If I manually publish a TwistStamped message to `/a300_0000/platform/cmd_vel`, then the rover moves. 

In the a200 simulation, with ROS2 Humble and Gazebo Fortress, the rover was controlled via `/a200_0000/cmd_vel` topic. Has this been changed to `/platform/cmd_vel` in ROS2 Jazzy and Gazebo Harmonic?

**To Reproduce**
Provide the steps to reproduce:
1. launch simulation: ros2 launch clearpath_gz simulation.launch.py
2. Change topic in Gazebo: `/a300_0000/cmd_vel`
3. Select "KEYBOARD"
4. Press W key
5. No movement
6. `ros2 topic echo /a300_0000/cmd_vel` shows data being published
7. Manually publish to /platform/cmd_vel topic: `ros2 topic pub /a300_0000/platform/cmd_vel geometry_msgs/msg/TwistStamped '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: "base_link"}, twist: {linear: {x: 0.5}, angular: {z: 0.1}}}'`
8. rover moves in simulation

tonybaltovski, 2025-04-28:
I tried to reproduce this but I could not.  Are you seeing any errors or warnings when launching?

daylanc, 2025-05-05:
Here's the output when I run `ros2 launch clearpath_gz simulation.launch.py`:

[INFO] [launch]: All log files can be found below /home/chad/.ros/log/2025-05-05-09-02-12-061977-chad-12100
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [12103]
[INFO] [parameter_bridge-2]: process started with pid [12104]
[INFO] [generate_description-3]: process started with pid [12106]
[parameter_bridge-2] [INFO] [1746471732.361551842] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[generate_description-3] Generated /home/chad/clearpath/robot.urdf.xacro
[INFO] [generate_description-3]: process has finished cleanly [pid 12106]
[INFO] [generate_semantic_description-4]: process started with pid [12128]
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[gazebo-1] [GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[generate_semantic_description-4] [INFO] [1746471732.786016418] [moveit_collision_updater.urdf]: Loaded a300-0000 robot model.
[generate_semantic_description-4] [INFO] [1746471733.027205843] [moveit_1017790300.moveit.core.robot_model]: Loading robot model 'a300-0000'...
[generate_semantic_description-4] [INFO] [1746471733.027235115] [moveit_1017790300.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[generate_semantic_description-4] [WARN] [1746471733.028009076] [moveit_1017790300.moveit.core.robot_model]: Link imu_0_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[generate_semantic_description-4] [INFO] [1746471733.028623869] [moveit_collision_updater.srdf]: Robot semantic model successfully loaded.
[generate_semantic_description-4] [INFO] [1746471733.040273942] [moveit_collision_updater.Self-Collisions]: Thread complete 36
[generate_semantic_description-4] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[generate_semantic_description-4]          at line 127 in ./src/class_loader.cpp
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Wood
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[... cut ...]
[gazebo-1] [Dbg] [SimulationRunner.cc:544] Creating postupdate worker thread (4)
[gazebo-1] [Dbg] [SimulationRunner.cc:544] Creating postupdate worker thread (5)
[gazebo-1] [GUI] [Wrn] [OBJLoader.cc:91] Both `d` and `Tr` parameters defined for "Chair". Use the value of `d` for dissolve (line 8 in .mtl.)
[gazebo-1] 
[gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2395] Triangle input semantic: 'COLOR' is currently not supported
[gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2395] Triangle input semantic: 'COLOR' is currently not supported

hilary-luo, 2025-05-06:
@daylanc If you install `ros-jazzy-clearpath-diagnostics` does this resolve the issue? I will look into fixing that error but I want to make sure that this resolves the issue you are seeing.

daylanc, 2025-05-06:
@hilary-luo installing the diagnostics package with 'sudo apt install ros-jazzy-clearpath-diagnostics' fixed the issue. I am now able to change the topic to '/a300_0000/cmd_vel' and move the robot. Thank you!!!!

stadekawa, 2025-05-08:
I am also having the same issue where the robot does not move when publishing to /a300_0000/cmd_vel. I see the same "package 'clearpath_diagnostics' not found error, so I tried to install ros-jazzy-clearpath-diagnostics, but got an "Unable to locate package" when trying to apt install. How can I install the package?

daylanc, 2025-05-22:
@hilary-luo I tried installing on a different laptop with a fresh Ubuntu 24.04 and ROS2 Jazzy and received the same issue as @stadekawa. I don't remember what I did on my original laptop to install this package successfully...
`sudo apt install ros-jazzy-clearpath-diagnostics`
Output:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-jazzy-clearpath-diagnostics

hilary-luo, 2025-05-22:
@stadekawa @daylanc Currently that package is hosted on the clearpath servers. My apologies that I haven't had a change to fix this yet. Use the following commands to install it:

```bash
wget https://packages.clearpathrobotics.com/public.key -O - | sudo apt-key add -
sudo sh -c 'echo "deb https://packages.clearpathrobotics.com/stable/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/clearpath-latest.list'
sudo apt update
sudo apt install ros-jazzy-clearpath-diagnostics
```

daylanc, 2025-05-22:
Thank you @hilary-luo. It worked perfectly for me. I can control the rover via /a300_00000/cmd_vel

closed 2025-06-09 by daylanc